mucamaca/vic-palacinke #2: GUI za raspberry pi

mucamaca, 2015-12-03:
Naredi grafičn interface za pi v JAVI. sej boš znal, a ne?

grizeldi, 2015-12-03:
Yes, sir!

grizeldi, 2015-12-05:
Ob izdelavi sem mimgrede naredu še framework za Matevža, ki pa bo mogu uporabt intellij , ker je sem uporabu pi plugin...

mucamaca, 2015-12-05:
Ja. Dobr. Sam ta tip mi ne odgovarja na maile. A mu napises na fb al pa 
mi das njegovo cifro, ce jo mas?

grizeldi, 2015-12-06:
Jaz ne imeti cifro ali fb biti...

mucamaca, 2015-12-06:
dobr indijanec

mucamaca, 2015-12-08:
@grizeldi a bi se dal narest tko, da je GUI fullscreen in da se nc okoli ne vidi okvirja od okna?
To bi blo super.
Nasu sm Matevza na githubu. Ogenjcek mu je ime

grizeldi, 2015-12-08:
Lahka nucam JWindow namest JFrame, če ti gre un border na vrh na jetra.

mucamaca, 2015-12-08:
Tocno to sm mislu :+1: . Zato, da na piju user ne bo mogu drezat po drugih stvareh ampak bo omejen na ta nas gui. Matevz si je naredu GitHub! ime mu je ogenjcek.(!)

mucamaca, 2016-01-09:
To lohk se mal pocaka

mucamaca, 2016-01-09:
Oh no, it's retarded!

mucamaca, 2016-03-06:
kako kej gre tale GUI?

mucamaca, 2016-04-30:
Meow

mucamaca, 2016-04-30:
Tole pa ni priority. Removite label.

mucamaca, 2016-05-26:
dej nared ze to!

grizeldi, 2016-05-26:
Ukin use teste, sprašvanja pa poročila za glasbo pa ti grem lahko takoj
delat.

mucamaca, 2016-05-26:
kaj pa pojektno pa antidepresive?

antif

grizeldi, 2016-05-27:
Kle mamo zdj se tima. Dve uri sva rabla ampak zdj mislm da prblizn zastop kako zadeve funkcionirajo.

mucamaca, 2016-05-27:
dobr.

grizeldi, 2016-06-03:
Prva varjanta je #32

mucamaca, 2016-06-27:
MEow
meOW

mucamaca, 2017-05-12:
No GUI - 
no issues with it.

closed 2017-05-12 by mucamaca