mucamaca/vic-palacinke #6: Pi side Arduino - Pi komunikacija

grizeldi, 2015-12-05:
Spisat del, ki bo komuniciral z arduinom in spremembe prenesel na GUI. Začetke (framework) sem že spisal, ostal bo naredil Matevž, ko si blagovoli naredit GitHub.

mucamaca, 2015-12-05:
Aha mamo zdej en issue brez assigneja. Ok

mucamaca, 2015-12-05:
Ja... @grizeldi tvoje navodilo, s cem se odpira pi GUI sm se mal poudaru. Da bo bl jasno... Zdej morm pa sam JDK na kompu zrihtat pa zagnat.

mucamaca, 2015-12-06:
Tal milestone mi je vsec

mucamaca, 2015-12-08:
@ogenjcek Dobu si prvi assignment. Bravo. Ce kej ni jasno, slisu sm, da mas rad javo, zato si dobil nalogo v javi. Nardil bomo user interface na raspberry piju in tvoja naloga je, da @grizeldiju pomagas narest pi program, konkretno tist del, ki komunicira z arduinom. Jaz pisem podobno stvar v Cju za arduino, tko da bova morala sodelovat. Komuniciranje gre pa prek 8 pinov, 4 v sako smer. Framework ti je ze @grizeldi napisu in je nekje v Pi_java_stuff folderju. Baje da mas rad eclipse, morm te razocarat, uporabljat bos moru IntelliJ IDEA ide z Pi pluginom. vse to pise na veliko v Pi_java_stuff, tko da upam, da bos znal. Ce cesa ne ves kar tukaj napisi vprasanje.

grizeldi, 2015-12-08:
@Ogenjcek ce rabs ksna navodila za uporabo kniznce pi4j sam povej.

mucamaca, 2016-03-06:
ta retard...

mucamaca, 2016-03-06:
mislm, da si ti to ze naredu @grizeldi , tko da lahk kr zaprem issue

closed 2016-03-06 by mucamaca