mucamaca/vic-palacinke #7: Mal popucat repozitorij

grizeldi, 2015-12-06:
Random fileje bi blo treba popredalčkat po folderjih in zradirat nepotrebno šaro.

grizeldi, 2015-12-06:
Sm hotu dodt labele, ampak mobiln github je dizaster.

mucamaca, 2015-12-06:
sploh nism vedu, da obstaja

grizeldi, 2015-12-06:
A lohk ti to nardis?
Sm te hotu assignat, pa na mobile ne dela.

mucamaca, 2015-12-06:
Ja bom

mucamaca, 2015-12-06:
Labelu sm ga nobenu se ne da. True story

grizeldi, 2015-12-06:
LOL
Sj b js nared sam nism doma, pol se morm pa nemo ucit.

mucamaca, 2015-12-06:
Good luck with nema. Bom pa ze jaz zrihtu
zapiram pa issueje ko nor. Tale bo ze drugi. Na sreco se jih da reopenat :8ball: dont ask me why

mucamaca, 2015-12-06:
No pa je

grizeldi, 2015-12-07:
ČŽŠ ji pa presledki sprobano delajo na malini, tko v nanotu kot če jih zalaufaš s pythonom, tko da pucanje se nadaljuje!

grizeldi, 2015-12-07:
Pa odloč se že kakšn bo ime, ka vsakič ka renamaš folder mi čist zjebe Ideo, pa morm na novo odpirat pa zapirat...

mucamaca, 2015-12-07:
Ok. Sam cist iz profesionalnosti in tega, da ce nimas default keymapa na slo na piju ne mors pisat not sumnikov poskusi ne met presledkov in sumnikov v file in folder namih

grizeldi, 2015-12-07:
Lahko pišeš šumnike in presledke. Sprobano. NWM pa če se da direktno, ampak prek Puttya se da

mucamaca, 2015-12-07:
putty sucks. use ssh command. Prosm lepo, dej brez sumnikov pa presledkov

mucamaca, 2015-12-07:
Zakaj mas pet folderjev enga v drugem v sourcu?

mucamaca, 2015-12-07:
Ful me ma, da bi to zbrisu.

grizeldi, 2015-12-07:
https://docs.oracle.com/javase/tutorial/java/package/namingpkgs.html

grizeldi, 2015-12-07:
Zj pa prosm nehi popravlat zadeve, ka niso pod tvojo domeno in ne veš čemu služjo.

closed 2015-12-07 by grizeldi

mucamaca, 2015-12-07:
too late

mucamaca, 2015-12-07:
sorry

mucamaca, 2015-12-07:
se dobr, da ma github commit history