mucamaca/vic-palacinke #13: A nardimo svojo kvazi main funkcijo al ne?

mucamaca, 2015-12-12:
To je bl predlog pa vprasanje kokr pa issue:
v glavnem programu vsakega arduino projekta morta bit void setup pa void loop.
funkcija setup se izvede na zacetku izvajanja in potem se zacne izvajat loop.
loop se potem laufa v neskoncnost k debil. Moj predlog je, da bi nardil funkcijo, k bi se enkrat poklicala v loopu in potem znotraj te nove funkcije bi nardil loop posebi.
Neki tazga:

```C
void setup(){
  // stuff 'n things 'n stuff
}
void loop_nadomestek(/*random argumenti*/){
  while(/*neki*/){
    // stuff, k bi bil dgac u loop()
  }
}
void loop(){
  loop_naodomestek();
  return;
}
```

zakaj?
zato, ker je precej bolj razumljivo in razvidno, da se tista stvar tm notr ponaula in tko mamo tut vec nadzora nad flowom programa

grizeldi, 2016-01-08:
Ne.

closed 2016-01-08 by grizeldi

mucamaca, 2016-01-08:
Zakaj ne?

grizeldi, 2016-01-09:
Ker je to najbl bedna ideja ever, in se jo lahko spomniš sam ti. Zakaj bi to rabu??? Argumente morš itak nekje dobit, in če jih odbiš v glavnem loopu je čist vseen če jih tut tm sprocesiraš. Upoštevi, da ima Arduino Mega samo 248 kB pomnilnika za program. Mislim da ga lahko bolj koristno ponucamo.

mucamaca, 2016-01-09:
1. Brez zamere, ampak v smislu prostora to zavzame popolnoma enako, ker po compilanju, asemblanju in whatever it does then to sploh ni vec funkcija. se posebi, ce compiler zna optimizirat.
2. void loop je strasansko annoying, ali pa vsaj motec, ce nisi motoricni ali kaksen drug debil al pa da se nikol nisi programiru prej. Pa se tole: uporablamo arduino uno in ima 32kiB (ti si pozabu na i) in to je vec kot zadost.

mucamaca, 2016-01-09:
Sej drgac, ce noces okej, sam predlog je bil. unih dveh pa itak ne bova vec sprasvala pa cakala, glede na to, kaj so do zdej nardil.

grizeldi, 2016-01-10:
Me figures out that @mucamaca has a voidloopphobia.

mucamaca, 2016-01-10:
ju not be so smart.
it's just annoying