mucamaca/vic-palacinke #14: Noben nic ne dela

mucamaca, 2015-12-25:
# Kva je zdej to????

mucamaca, 2015-12-25:
@grizeldi @ogenjcek @MaticJeCar Nic ne delate, ze dva tedna ni nobenga commita razen mene pa grizeldija sploh *nic?!?!?!?!?!*

grizeldi, 2015-12-26:
@mucamaca razlog je enostaven. Capudra ni nikjer in ostale ekipe tudi nič
ne delajo. Meni se zdi brezveze neki matrat za nekaj kar potem sploh ne bo
uporabljeno. Naj najprej ostali kaj naredijo, potem pa se spet pridružujem
programiranju. BTW: Uspoosbl smo 3D printer. Kind of.

mucamaca, 2015-12-27:
> Links:

A bo kdo mogoč za težil na fb skupini??

K če ne bo, pol očitno res ne bo nc.

2015-12-26 17:04, je grizeldi napisal

mucamaca, 2016-01-09:
To sploh ni issue. To je vzdih, ki ni izpolnil svojega namena in poslanstva.
Najboljs, da ga closam

closed 2016-01-09 by mucamaca