mucamaca/vic-palacinke #16: Pogruntat protokol za serial komunikacijo

grizeldi, 2016-01-10:
Treba je izumit katera koda bo kaj pomenila in to zapisat v en txt.
Pomoje je najbolje da ima en paketek (sporocilo) 3 bajte:
1. bajt: koda sporocila
2. bajt: prvi del podatka
3. bajt: drugi del podatka

V tist txt pa sam napismo kaj je kjera koda pa kaj pomeni podatek.

mucamaca, 2016-01-10:
zakaj prvi in drugi del podatka?
mislm da bi bil en bajt za podatek cist zadost.
@grizeldi razlozi mi prosim

mucamaca, 2016-01-10:
Bom jez to naredu

grizeldi, 2016-01-10:
Si kdaj pomislu da je lohk podatek vecji od 255 (oziroma 127 v primeru
jave)?

mucamaca, 2016-01-10:
ja, sam se ne spomnem nobenga tazga podatka, ki bil relevanten za tele 
palacinke

grizeldi, 2016-01-10:
Obrnjenost stepper motorja (ma 400 korakov).

mucamaca, 2016-01-10:
ok, sam zakaj bi to posiljal piju pa tut lahko delis z dva pa bo se zmer 
blazno natancno

mucamaca, 2016-01-10:
se kej?

grizeldi, 2016-01-10:
To ma point. Ampak lahko pa pol posles dva podatka z eno komando.

mucamaca, 2016-01-10:
to je nesmiselno.
pol ne ves, kaj pomen ta drugi podatek
itak te pa nc ne stane, ce posles pol dve komandi po en podatek.
pa se racuni, da je en arduino integer 16 bitov, tko da se ena komanda z enim podatkom lih lepo stisne not

grizeldi, 2016-01-10:
Oki. Cr bo treba itak lahko naknadno spremenimo nazaj na 3 bajte.

mucamaca, 2016-01-10:
prav.
mal off-topic: v kirem programu se nardi 3d model, ki ga das cnc-ju da izrezka?

mucamaca, 2016-01-10:
jaz moram it, ker gremo druzinsko tect. adijo.
cez uro pa pol bom nazaj

mucamaca, 2016-01-10:
AAnd im back

grizeldi, 2016-03-06:
A si ze naredu?

mucamaca, 2016-03-06:
ne

mucamaca, 2016-03-11:
yu is maca

mucamaca, 2016-03-11:
ja

mucamaca, 2016-03-13:
koncni izdelk je u serial-protokol.txt
ce komu kej ni prov naj pove.

closed 2016-03-13 by mucamaca