mucamaca/vic-palacinke #35: Popravit in dopolnit arduino debug filter, da bo skladen MeWP protokolom

mucamaca, 2016-06-27:
protokol je definiran v MeWP specifikacija.txt

grizeldi, 2016-06-27:
Bullshit. Najpri zrihti ime pa specifikacijo tko da bo smiselno, pol pa bom kej naredu.

mucamaca, 2016-06-27:
ja ni se koncano.

mucamaca, 2016-06-27:
a prides jutr na zavod?

mucamaca, 2016-06-27:
a ves da je bil rms enkrat v Sloveniji?

grizeldi, 2016-06-28:
Kill it.

mucamaca, 2016-06-28:
ni ga vec v sloveniji. poleg tega je prov car.

mucamaca, 2016-06-28:
a ves da sem napisu awesome linux device driver, ki se imenuje /dev/meow in vsakič ko poskušaš brat z njega prebereš meow.

mucamaca, 2016-06-28:
zdej ni več neuporaben

grizeldi, 2016-06-28:
Kje se naštela na githubu da enga ignoraš?

mucamaca, 2016-06-28:
A ti šalca teži al kaj?

mucamaca, 2016-06-28:
A ves, da ima ta issue ze celih 11 kommentov

grizeldi, 2016-06-28:
Po novem sm te blokiru.
Ko bos meu kej pametnga za povedat mi pa posl mejl. Ce mi bos pa se tm sranje posilju te bom pa se tm u spam prestavu.

mucamaca, 2016-06-28:
Misl da salca ne gleda tega, tko da mu lahk ti posles mail

mucamaca, 2016-06-28:
no, zdaj mislm, da je vse skupaj v redu napisan. se preimenoval sem tist file.
btw linux kernel  4.0 je bil codenamed Hurr Durr ima a sheep.

mucamaca, 2016-06-28:
4.7 je pa Psychotic stoned sheep.

grizeldi, 2016-06-29:
Jamal, you is stupid!
Blokiral sem uporabika @mucamaca.
In tist gor je tut veljal teb. Bolnih for pačni treva quotat.

mucamaca, 2016-08-29:
Ja vem.

mucamaca, 2016-11-26:
mjau!

closed 2016-11-26 by mucamaca